yzyhk904/USB_SampleRate_Changer #2: Please add options like remote_submix and internal audio recording

iori31, 2021-09-16:
Please add options like 'remote_submix' and internal audio recording(need root) with bit width and sample rate. Thus we can transfer whole system audio stream to our remote player by DLNA protocol when we playing hi-res audio in Apple Music.

I've tried but 'remote_submix' parameters in config files didn't work. Moreover, I don't know how to change internal audio recording's bit width and sample rate.

yzyhk904, 2021-09-16:
r_submix can accept 48kHz 16bit or 44.1kHz 16bit only. You can transmit 44.1kHz 16bit audio by airMusic or google cast (mirroring to AV receivers) bit-perfectly now. If you specify 48kHz (default setting), airMusic resamples 48kHz audio input down to 44.1kHz internally (maybe 44.1kHz -> 48kHz -> 44.1kHz; resamples twice).

I think the best is setting the configuration 44.1kHz 16bit for r_submix.

My recommendation is using "hifi maximizer" and Youtube music or amazon music unlimited. I think the audio quality of AAC 256kbps on "hifi maximer" is the same as Apple Music Hires.

iori31, 2021-09-17:
> r_submix can accept 48kHz 16bit or 44.1kHz 16bit only. You can transmit 44.1kHz 16bit audio by airMusic or google cast (mirroring to AV receivers) bit-perfectly now. If you specify 48kHz (default [...]

Wow! Is that a hardware? I'll try it.

yzyhk904, 2021-09-17:
"hifi maximizer"? It is a magisk module in my another repository.

iori31, 2021-09-18:
> "hifi maximizer"? It is a magisk module in my another repository.

Great! Now it is running on device.

closed 2021-10-20 by yzyhk904