yzyhk904/USB_SampleRate_Changer #3: Script outputs an error

ronoren, 2021-09-16:
Hello. 
My device is Samsung Galaxy S10e and it is rooted with Magisk.

I executed this command in Termux:
```bash
su -c ./USB_SampleRate_Changer/USB_SampleRate_Changer.sh --auto
```
and the output was:
```
Can't find service: media.audio_policy
audioserver is not running!
```

The samplerate also didn't change when I checked it with `dumpsys-filtered.sh`

I know that your script works on other devices as I teseted it also on OnePlus 6. But what happend on this Galaxy?
Do you know what should I do to resolve it?
Thanks for the awsome module!

yzyhk904, 2021-09-16:
"dumpsys media.audio_policy" command is used for getting where the audio_policy configuration XML file is.
The message reported this command failed and /vendor/etc/audio_policy_configuration.xml was assumed.
And I assume that the state of the audioserver can be obtained by "getprop init.svc.audioserver" command.

I guess the audioserver on S10e is not a system service, but a vendor service or having a strange name.
Anyway, the success of "dumpsys media.audio_policy" command execution is needed.

ronoren, 2021-09-16:
Using the commands that you mentiond I figured out that the reason that the error was showing is that i executed the script with the `--auto` argument right after I executed the command with the `--reset` argument. I noticed that the `getprop init.svc.audioserver` command returned `restarting` and after a couple of seconds it returned `running`.

So the script doesn't return an error anymore as it was a matter of waiting for the service to restart.
However, it still doesn't change the sample rate to the song's sample rate. It locks the sample rate from 48000 to 44100. I even tried to change the sample rate to a constant one with this command:
```bash
su -c ./USB_SampleRate_Changer/USB_SampleRate_Changer.sh 192k 24
```
but still it locked on 44100 and 32 bit.

I'm providing also the outputs of `dumpsys media.audio_flinger` and `dumpsys media.audio_policy` when I played a song with 192kHz using "UAPP" (USB Audio Player Pro) after I executed the script with the `--auto` argument.
[audio_flinger.txt](https://github.com/yzyhk904/USB_SampleRate_Changer/files/7180500/audio_flinger.txt)
[audio_policy.txt](https://github.com/yzyhk904/USB_SampleRate_Changer/files/7180501/audio_policy.txt)

yzyhk904, 2021-09-17:
First of all, "--reset" option means restarting the audioserver and even audio hal servers. This takes a while. You don't need "--reset" option unless your device is in a trouble. Usual the changer executes restarting the audioserver only (this takes a second or so).

The changer doesn't change the sample rate automatically (but default is 44.1kHz and 32bit depth). From your attached info, the changer investigated your device and determined to use "safe" mode (44.1kHz and 32bit is fixed for 3.5mm jack and internal speakers while specified sample rate and bit depth is used for a USB DAC). In this mode, you can change the sample rate and bit depth only for a USB DAC (using a standard USB Audio Class driver, but not a UAPP's proprietary internal driver).

Don't you use a 3.5mm jack? try ``su -c ./USB_SampleRate_Changer/USB_SampleRate_Changer.sh --legacy  192k 32``. It may work.

yzyhk904, 2021-09-23:
> Can't find service: media.audio_policy
> audioserver is not running!

This situation was reproduced on my environment if I used ArrowOS 11.0, a specific USB DAC, UAPP (using the hardware volume adjustment of the DAC only) and a usual music player app together. This situation doesn't happen without the driver initializing of UAPP.