yzyhk904/USB_SampleRate_Changer #6: Sample rate changed at wrong device

CKH0, 2022-03-08:
[log.txt](https://github.com/yzyhk904/USB_SampleRate_Changer/files/8203028/log.txt)
This log includes dumpsys-filtered.sh result 
My DAC show that it isn't outputting at 96 kHz 
then I checked it seems like it changed the sample rate but on a different output device

yzyhk904, 2022-03-08:
I checked your log. If you executed `USB_SampleRate_Changer.sh --auto` exactly, it changed the sample rate of a 3.5mm jack, an internal speaker and USB DAC's to "44.1kHz & 32bit" mode. `--auto` doesn't mean automatic adaptation to the outputting frequency of a music player, but automatic investigation of your device to choose the mode of audio_policy_configuration.xml, i.e., offload, bypass-offload, safe, or safest.

I guess you would like to use the Apple music app and select the best settings. If so, I recommend executing `USB_SampleRate_Changer.sh 96k 32`. and `extras/change_resampling_quality.sh 160 480 91`. And see the description of "extras/jitter-reducer.sh" .

CKH0, 2022-03-08:
so how can i set this tool tp bypass SRC :thinking:

yzyhk904, 2022-03-08:
You can change the sampling frequency appropriately by hand each time before your player changes the outputting sampling frequency. But the above resampling quality "160 480 91" means a mastering quality and you cannot hear any resampling distortion even with SRC, perhaps.

I think "bit-perfect" isn't important, but distortion is truly important.

closed 2022-03-14 by yzyhk904