yzyhk904/USB_SampleRate_Changer #9: Feature request

akavkl, 2022-05-08:
Could you add float bit depth. Here to test in needed.

yzyhk904, 2022-05-08:
I added an option for the float bit depth in the latest release.

I don't have any device with float output capability, so Android "audioserver" stops or doesn't route audio data if I specify the float option.

You have a Pixel 6 or something, don't you?  Can you check this release?

akavkl, 2022-05-08:
I have OnePlus 6 and Zuk z2 to test with and both are float capable so I'll try

akavkl, 2022-05-10:
Output for Non-USB
https://pastebin.com/b2wKmQiS
Expected output for Non-USB
https://pastebin.com/fEf0SYgv

yzyhk904, 2022-05-11:
What did you do?

`sh /sdcard/USB_SampleRate_Changer/USB_SampleRate_Changer.sh --offload-hifi-playback 384k float`

Did you execute the above command? "Output for Non-USB" seemed to be in safest mode.
Probably, my script determined "safest mode" to be the best mode for your device. In this mode internal outputs are fixed to be 48kHz & 16bit and a USB output only varies on your request.

In "offload-hifi-playback" mode you probably cannot use bluetooth earphones and USB DAC's. The right option seems to be "--legacy 384k float".

akavkl, 2022-05-11:
sh ./USB_SampleRate_Changer.sh --offload-hifi-playback 384k float
is the command I used from script's directory, but Bluetooth audio was working fine, was on LDAC btw

akavkl, 2022-05-15:
sh /sdcard/USB_SampleRate_Changer/USB_SampleRate_Changer.sh --offload-hifi-playback 384k float
Float worked but still in 48khz

yzyhk904, 2022-05-16:
The audioserver on your phone would preserve 48kHz while trying to change the sample rate. This is usual on old devices, but doesn't happen on recent Qcomm devices.

I think you device's native format for internal speakers is fixed at 48kHz & 24bit (or 32bit). You can see the actual format by executing "extras/alsa-hw-params.sh" while playing music. Probably "float" is virtual (not native format).

akavkl, 2022-05-16:
Output from dumpsys media.audio_flinger from my go-to script:-

`Output thread 0x73a84f1a60, name AudioOut_D, tid 3074, type 0 (MIXER):
  Sample rate: 384000 Hz
  HAL format: 0x5 (AUDIO_FORMAT_PCM_FLOAT)
  Output devices: 0x2 (AUDIO_DEVICE_OUT_SPEAKER)
  Timestamp stats: n=20625 disc=75 cold=1 nRdy=0 err=9 rate=0.999964 jitterMs(ave=0.0828951 std=2.04666 min=-37.3715 max=75.4905) localSR(383977, 3.61539e-09) correctedJitterMs(ave=0.0856429 std=2.04014 min=-37.235 max=75.4905)
  AudioStreamOut: 0x7152b7a5b0 flags 0x6 (AUDIO_OUTPUT_FLAG_PRIMARY|AUDIO_OUTPUT_FLAG_FAST)
  Local log:
   05-16 18:12:47.913 AT::add       (0x7212b89770)         564     no   3471    1865    1062 A  0x000 00000001 00000001  48000  5   5  4  -inf     0     0     0  00000000  24000       0 f         0        0       new
   05-16 19:01:56.347 AT::add       (0x7212b86a60)         569     no   3471    1889    1075 A  0x000 00000001 00000001  48000  5   5  4  -inf     0     0     0  00000000  24000       0 f         0        0       new
   05-16 19:42:50.513 AT::add       (0x7212b816b0)         578     no   3471    1985    1098 A  0x000 00000001 00000001  48000  5   5  4  -inf     0     0     0  00000000  24000       0 f         0        0       new
   05-16 19:46:07.380 AT::add       (0x7212b836e0)         582     no   3471    2049    1108 A  0x000 00000001 00000001  48000  2   6  4  -inf     0     0     0  00000000  24000       0 f         0        0       new
   05-16 19:46:33.165 AT::add       (0x7212b836e0)    S    583     no  16861    2057    1109 A  0x000 00000001 00000001  44100  1   d  4  -inf     0     0     0  00000000   4354    4354 F         0        0       new
   05-16 19:49:31.521 AT::add       (0x7212b89100)         584     no   3471    2065    1110 A  0x000 00000001 00000001  48000  5   5  4  -inf     0     0     0  00000000  24000       0 f         0        0       new
   05-16 20:44:54.741 AT::add       (0x7212b89770)         596     no   3471    2161    1130 A  0x000 00000001 00000001  48000  5   5  4  -inf     0     0     0  00000000  24000       0 f         0        0       new
   05-16 21:02:10.397 AT::add       (0x7212b83070)         628     no   3471    2361    1210 A  0x000 00000001 00000001  48000  5   5  4  -inf     0     0     0  00000000  24000       0 f         0        0       new

Output thread 0x73a8365a60, name AudioOut_1D, tid 3082, type 0 (MIXER):
  Sample rate: 384000 Hz
  HAL format: 0x5 (AUDIO_FORMAT_PCM_FLOAT)
  Output devices: 0x2 (AUDIO_DEVICE_OUT_SPEAKER)
  Timestamp stats: n=464114 disc=386 cold=0 nRdy=0 err=40 rate=0.999999 jitterMs(ave=0.000256464 std=0.193674 min=-79.9773 max=189.996) localSR(384002, 3.57197e-09) correctedJitterMs(ave=-4.19029e-05 std=0.00447693 min=-1.59187 max=189.991)
  AudioStreamOut: 0x7152b77130 flags 0x8 (AUDIO_OUTPUT_FLAG_DEEP_BUFFER)
  Local log:
   05-16 20:43:46.015 AT::add       (0x7212b7f680)         591     no  29434    2081    1121 A  0x000 00000001 00000003  48000  3   1  2  -inf     0     0     0  00000000  12000   12000 f         0        0       new
   05-16 20:44:03.534 AT::add       (0x7212b82a00)         593     no  29434    2081    1125 A  0x000 00000001 00000003  48000  3   1  2  -inf  -inf  -inf     0  00000000  12000   12000 f         0        0       new
   05-16 20:44:05.994 AT::add       (0x7212b83070)         594     no  29434    2081    1126 A  0x000 00000001 00000003  48000  3   1  2  -inf -0.99 -0.99     0  00000000  12000   12000 f         0        0       new
   05-16 20:44:16.875 AT::add       (0x7212b83070)         594     no  29434    2081    1126 r  0x000 00000001 00000003  48000  3   1  2  -inf -0.99 -0.99     0  0007BC00  12000   12000 f         0        0  375.41 k
   05-16 20:44:58.313 AT::add       (0x7212b83070)         597     no  29434    2081    1131 A  0x000 00000001 00000003  48000  3   1  2  -inf     0     0     0  00000000  12000   12000 f         0        0       new
   05-16 20:55:26.488 AT::add       (0x7212b87740)         622     no  29434    2081    1204 A  0x000 00000001 00000003  48000  3   1  2  -inf     0     0     0  00000000  12000   12000 f         0        0       new
   05-16 20:55:52.386 AT::add       (0x7212b7f680)         623     no  29434    2081    1205 A  0x000 00000001 00000003  48000  3   1  2  -inf     0     0     0  00000000  12000   11520 f         0        0       new
   05-16 20:55:54.203 AT::add       (0x7212b83070)         624     no  29434    2081    1206 A  0x000 00000001 00000003  48000  3   1  2  -inf     0     0     0  00000000  12000   12000 f         0        0       new
   05-16 20:55:55.583 AT::add       (0x7212b863f0)         625     no  29434    2081    1207 A  0x000 00000001 00000003  48000  3   1  2  -inf     0     0     0  00000000  12000       0 f         0        0       new
   05-16 20:56:03.829 AT::add       (0x7212b863f0)         625     no  29434    2081    1207 r  0x000 00000001 00000003  48000  3   1  2  -inf     0     0     0  0003C780  12000   12000 f         0        0  349.74 k
   05-16 20:57:00.217 AT::add       (0x7212b83070)         626     no  29434    2081    1208 A  0x000 00000001 00000003  48000  3   1  2  -inf  -inf  -inf     0  00000000  12000   12000 f         0        0       new
   05-16 20:57:03.598 AT::add       (0x7212b83070)         627     no  29434    2081    1209 A  0x000 00000001 00000003  48000  3   1  2  -inf  -inf  -inf     0  00000000  12000   12000 f         0        0       new
   05-16 22:09:40.762 AT::add       (0x7212b863f0)         629     no  24693    2369    1211 A  0x000 00000001 00000003  44100  3   1  2  -inf  -1.2  -1.2     0  00000000  11025       0 f         0        0       new

`

yzyhk904, 2022-05-17:
Any problem? You showed that the audioserver set a mixer at 384kHz float. This format is virtual, not a native format. Audio data is transferred from a music player via an OS mixer (with resampling) via a tiny ALSA driver (with resampling) to a physical audio device.  Only ALSA devices show the final (native) audio format.

There are some phones that wouldn't change its final audio format (i.e., its ALSA driver resamples any input audio data in  its fixed native audio format, say 48kHz 24bit, or hangs up at all). Phones adopted Qcomm recent Snapdragon usually varies its native audio format as the audioserver requests except float and more than 384kHz.

closed 2022-06-13 by yzyhk904