yzyhk904/USB_SampleRate_Changer #10: Running USB_SampleRate_Changer.sh jitter-reducer Using ViPER4Android FX to adjust volume with delay Android 11 Redmi Note 9 MTK

a6886126, 2022-06-12:
Running USB_SampleRate_Changer.sh jitter-reducer Using ViPER4Android FX to adjust volume with delay Android 11 Redmi Note 9 MTK

yzyhk904, 2022-06-13:
Since USB_SampleRate_Changer.sh would restart the audioserver (OS mixer) even while playing music via Viper4AndroidFX, your running music player and Viper4AndroidFX may possibly unadaptable to the restart. Try execute USB_SampleRate_Changer.sh and jitter-reducer.sh before playing music (and connecting bluetooth) or after quitting the music player (and disconnecting bluetooth).

By the way, in my opinion I think Viper4Android isn't needed if you use my magisk module "audio-misc-settings" (using a mastering quality resampler, a 100 step volume, etc.,) because Viper4Android only distorts music under low distortion situations.

a6886126, 2022-06-13:
> 

> Since USB_SampleRate_Changer.sh would restart the audioserver (OS mixer) even while playing music via Viper4AndroidFX, your running music player and Viper4AndroidFX may possibly unadaptable to the [...]

> 由于USB _ samplate _ changer．sh即使在通过Viper4AndroidFX播放音乐时也会重新启动audioserver（os混频器），因此正在运行的音乐播放器和Viper4AndroidFX可能无法适应重新启动。尝试执行USB _ samplate _ changer．sh和jitter－reducer．sh之前播放音乐（和连接蓝牙）或退出音乐播放器（和断开蓝牙）。
> 
> 顺便说一下，在我看来，如果你使用我的magisk模块“audio－misc－settings”（使用掌握质量重扩音器，100步音量等），我认为Viper4Android是不需要的，因为Viper4Android只在低失真情况下扭曲音乐。

> Since USB_SampleRate_Changer.sh would restart the audioserver (OS mixer) even while playing music via Viper4AndroidFX, your running music player and Viper4AndroidFX may possibly unadaptable to the [...]

No sound when using audio-misc-settings phone

a6886126, 2022-06-13:
How can I make the sound not delayed?

yzyhk904, 2022-06-13:
Did you uninstall Viper4AndrodFX completely before installing "audio-misc-settings"? As "audio-misc-settings" would disable the framework interfacing to all equalizers and other effects, Viper4AndroidFX cannot function or works unintendedly.

I guess the reason of the delay is probably that Viper4AndroidFX cannot adapt to the restart of the audioserver. So my answer is to try uninstalling Viper4AndroidFX completely.

a6886126, 2022-06-13:
> 

I use Viper4AndrodFX to calibrate and equalize my headphones Viper4AndrodFX is very useful Viper4AndrodFX can also simulate the sound of other headphones with autoeq  https://github.com/jaakkopasanen/AutoEq

a6886126, 2022-06-13:
> Did you uninstall Viper4AndrodFX completely before installing "audio-misc-settings"? As "audio-misc-settings" would disable the framework interfacing to all equalizers and other effects, [...]

https://github.com/jaakkopasanen/AutoEq

a6886126, 2022-06-13:
> Did you uninstall Viper4AndrodFX completely before installing "audio-misc-settings"? As "audio-misc-settings" would disable the framework interfacing to all equalizers and other effects, [...]

I use Viper4AndrodFX to calibrate and equalize my headphones Viper4AndrodFX is very useful Viper4AndrodFX can also simulate the sound of other headphones with autoeq   
https://github.com/jaakkopasanen/AutoEq

a6886126, 2022-06-13:
--auto --reset How to use these directives? where to run I wouldn't use

yzyhk904, 2022-06-14:
In my opinion, Viper4Android is useless at all under no resampling distortion and no DRC situations or for hi-fi audio use.

Did you read the README?

closed 2022-07-02 by yzyhk904