yzyhk904/USB_SampleRate_Changer #11: Syntax error: "(" unexpected on termux.

maniaossplayer, 2022-06-14:
~ $ sh /sdcard/USB_SampleRate_Changer/USB_SampleRate_Changer.sh --auto
/sdcard/USB_SampleRate_Changer/USB_SampleRate_Changer.sh: 7: /sdcard/USB_SampleRate_Changer/functions3.shlib: Syntax error: "(" unexpected
~ $

yzyhk904, 2022-06-14:
I cannot reproduce this error. Did you use the latest version?
If so, try another terminal emulator. I recommend using "sManager".

maniaossplayer, 2022-06-15:
With sManager same error occurs

yzyhk904, 2022-06-16:
I cannot reproduce this error till now while investigating old released versions. Do you really use the latest released version (i.e., ["USB_SampleRate_Changer-V2-5-0-1.zip"](https://github.com/yzyhk904/USB_SampleRate_Changer/releases/download/V2.5.0/USB_SampleRate_Changer-V2-5-0-1.zip)) ?

maniaossplayer, 2022-06-16:
Of course I'm using the latest version

yzyhk904, 2022-06-16:
I guess your scripts are corrupted. The syntax error cannot happen without corruptions because the scripts here don't have any syntax errors.

Please download the latest version again.

maniaossplayer, 2022-06-16:
I have re-downloaded the script 4 times, and the error still occurs.

yzyhk904, 2022-06-16:
I searched such errors on the internet. Some suggests that this occurs from the difference between bash and ash. My scripts are written for ash (in Busybox or Toybox) that is standard on the Android OS.

Do you use bash instead of ash?

maniaossplayer, 2022-06-16:
how to change to ASH shell?

yzyhk904, 2022-06-16:
I don't know. You changed the shell when installing Termux, didn't you?

maniaossplayer, 2022-06-16:
I didn't change anything, I want to try reinstalling busybox.

maniaossplayer, 2022-06-16:
I think it's a bug from magisk 25.0 on my device, I downgrade the magisk and it's work flawlessly, thanks for you help.

closed 2022-06-16 by maniaossplayer