yzyhk904/USB_SampleRate_Changer #12: Not perfekt.

faakarna, 2022-08-15:
Hi. Thanks for your great script. 
Having some trouble with my Hiby W3 Saber USB DAC, which somehow suddenly wont Play "bit perfect" in uapp anymore. This means that the Usb DAC only plays in 44.1/48 kHz, which really annoys me. Have been using All og your great magisk modules for a Long time, which really improves audio quality in my Samsung A53 5g. 

I have been playing around with the Usb_samplerate_changer script now, to see, if there is any possibility to make my devices Play in bit perfect mode again/adaptive, so im not stuck in 44/48 kHz samplerate.
Could you help me, in figuring out which settings to use, in your script, if i would like this feature? Would really appreciate that. 

(In Hiby Music app, it says that peripheals not support more than 48 kHz, which really hasnt been any problem before).

yzyhk904, 2022-08-15:
First of all, I cannot understand what you've done with A53 5G and the DAC. Did you update my script, UAPP or A53's firmware? My understanding is that UAPP can control your DAC via its own USB driver and use "bit perfect" mode without my script.

Any way, I think you can enjoy music by a music streaming app with "USB_SampleRate_Changer", "audio-misc-settings", and "audio-jitter-silencer" without resampling distortion. "Bit perfect" is meaningless under no resampling distortion environment by a mastering quality resampler.

Can you use scripts under "extras" folder? By "dumpsys-filtered.sh" you can view the state of the OS mixer (AudioFlinger) while playing music. Please check the "Sample rate" and "HAL format".

faakarna, 2022-08-15:
Sorry. I think i have found the reason why my dac suddenly gets Limited to 48khz/16. Its because the dac(w3 Saber) somehow has a limit of 48/16. Just really weird why its been playing flawless in uapp in bit perfect, in 96khz 24bit hi ress files. I simply chose to buy a new dac/Amp. Tested my old Dragonfly black, and It was processing in 96/24, so it MUST be the device i guess. Sorry for my the interruption, but i was sure, IT was a limit of the phone, that didnt want to adapt the sample rate and the depht. Thanks for your great magisk modules, and have a nice evening.

yzyhk904, 2022-08-16:
There is a script "extras/alsa-hw-params.sh" in "USB_sampleRate_Changer" that can show the USB detailed profile of your DAC while playing music. You can obtain the sample rate capabilities of your DAC. I guess your DAC has a switch between UAC v1 (44/48 kHz) and UAC v2 (96/192 kHz).

closed 2022-09-07 by yzyhk904