yzyhk904/USB_SampleRate_Changer #13: Echo during voice call using earpiece

OmarAbugazia, 2022-09-07:
Hi. Thanks a lot for this script.
Whenever I change the sample rate for usb, all phone calls have an echo on the other end .
I would appreciate it if this issue could be fixed cause I like this script.

yzyhk904, 2022-09-07:
What device and OS are you using? I've heard stuttering of phone call till now, but not any echo.
There are audio policy configuration templates in my script folder "templates". These are for least common settings, so you may need some modifications for them.

OmarAbugazia, 2022-09-07:
> What device and OS are you using? I've heard stuttering of phone call till now, but not any echo.
> There are audio policy configuration templates in my script folder "templates". These are for least common settings, so you may need some modifications for them.

OnePlus 7 pro, Android 12 ( Flamingo custom rom)

What modifications can I try?

yzyhk904, 2022-09-08:
Do you use "bypass-offload" mode (default for Qcomm devices)? If so, you've changed the usb dac samplerate (and bit depth) and also simultaneously those of speaker and earpiece.

I guess your device cannot handle phone calls with bit depth 32 or 24.  Can you test phone calls after setting 48kHz 16bit mode and 96kHz 24bit mode separately on your device?

OmarAbugazia, 2022-09-09:
Yes I use bypass offload ( I think that's better for usb dac right? ) 

Using --reset to revert back to the default 48khz makes phone calls normal again, is there anyway to only change the audio settings for usb without affecting the speaker and earpiece? 

And no I use a samsung dongle that supports 192khz and 32 bit maximum.

yzyhk904, 2022-09-10:
Can you test "USB_SampleRate_Changer.sh --bypass-offload 96k 24" ? I think this will work.
If this doesn't work, try "USB_SampleRate_Changer.sh --safest 96k 32". "--safest" mode doesn't change the samplerate of speakers and earpieces but does that of a usb DAC. (In this mode, you cannot use bluetooth devices)

OmarAbugazia, 2022-09-11:
> Can you test "USB_SampleRate_Changer.sh --bypass-offload 96k 24" ? I think this will work.
> If this doesn't work, try "USB_SampleRate_Changer.sh --safest 96k 32". "--safest" mode doesn't change the [...]

First command didn't work but second one "--safest " did. No more echo in calls!
Thanks for the fix!

closed 2022-09-13 by yzyhk904

yzyhk904, 2022-09-20:
I reproduced the same issue here with Xiaomi POCO F3 (Android 12, but not 11). I think the issue has been fixed with V2.5.4. And I added "--bypass-offload-safer" mode for keeping internal speakers to be 48kHz & 16/24bit when using more than 384kHz USB modes.