yzyhk904/USB_SampleRate_Changer #14: USB DAC device number switching?

r3muxd, 2022-09-14:
Sorry if this is off-topic, but I can't find anything else about this online. The SteelSeries Arctis 7 headset transmitter, when connected over USB, shows a "Chat" device (shows up first, 16000hz max, only mono) and a "Game" device (shows up last, full quality audio, stereo). Android only lets me use the first device seemingly no matter what - the only thing that I've found works is running alsa_aplay as root with the "hw:0,1" syntax. Is there any alsa or prop tweak I could apply to let me use the other device?
[image]
It shows up as this in /dev. pcmC0D0p is chat, pcmC0D1p is game (I need a way to switch the device number).

yzyhk904, 2022-09-14:
In "extras" folder there are two scripts "alsa-hw-params.sh" and "dumpsys-filtered.sh". First of all, try these scripts showing an ALSA device profile and an audio routing info of the OS mixer while playing music. Then you can get the actual capability of your device (perhaps, about "Chat" port).

Android has two device types for USB devices, i.e., "AUDIO_DEVICE_OUT_USB_HEADSET" (probably "Chat"; this is usually for music play back and voice communication) , "AUDIO_DEVICE_OUT_USB_DEVICE" (possibly "Game").

r3muxd, 2022-09-17:
https://termbin.com/iafn
Also, just to clarify, the "Chat" and "Game" devices are different devices on the headset (it has a physical slider to mix audio from the chat and game devices, and the two devices show up in Windows as separate.)

yzyhk904, 2022-09-17:
The above log shows that your phone outputs music by a hardware offload USB driver with 48kHz/16bit mono mode and "Chat" is a sound card device "/proc/asound/card1" on the Android OS. Probably, "Game" is another sound card device "/proc/asound/card2". If this true, you can see its profile by opening "/proc/asound/card2/stream0".

Can you deactivate "Chat" by any mean? Then the OS mixer will change the audio route to "Game". I guess "Chat" and "Game" are both active now and "Chat" has priority.
(Shortly your phone is connected to two DAC dongles with headphones both simultaneously.)

I think there is no way to let the OS mixer switch from "Chat" to "Game" without deactivating "Chat" (disconnecting headphones from "Chat" only).

r3muxd, 2022-09-17:
I would do that, but I can't figure out how.

yzyhk904, 2022-09-18:
Is there any mode switch on its transmitter? Unless so, do you like to connect a line-out (or via a DAC dongle) of your phone to the line-in of the transmitter?

r3muxd, 2022-09-18:
The transmitter has no mode switch and is attached only over one USB, sadly.

closed 2022-10-06 by yzyhk904