yzyhk904/USB_SampleRate_Changer #16: Questions about applying to bluetooth headsets and ...

Ed20010424, 2022-11-17:
When I try to run 
`sh /sdcard/USB_SampleRate_Changer/USB_SampleRate_Changer.sh --bypass-offload-safer 96k 24`, 
the Bluetooth headset no longer emits sound, but instead emits sound from the phone. What should I do?
My Bluetooth headset supports LHDC_V4 and has 24/96kHz turned on in the developer options, and is connected at the optimized audio quality (900kbps).
Another question, when I use a DAC connected to a wired headset and set to 384k 32, will there be a loss of audio quality when playing audio below 384k? For example 44.1 or 88.2kHz?
This question comes from previous knowledge , when playing audio, dumpsys media.audio_flinger log will appear in the corresponding sampling rate, that is, adaptive, in the Xiaomi device MIUI, Xiaomi gives a solution to avoid resampling, when playing music through the mediaplayer framework built player, MIUI will enable a separate OFFLOAD channel and adapts to the sampling rate. But unfortunately, the streamers don't use mediaplayer framework.
I'm sorry to ask you these questions, my English is not very good, maybe the readme mentions them and I may not understand them correctly.
Finally, thank you for your work.

yzyhk904, 2022-11-18:
`bypass--offload-safer` changes the sample rate of a mixer for USB only and disables "a2dp hardware offload" for bluetooth headsets so that the sample rate of a mixer for bluetooth can be changed dynamically by specifying in the developer options. In short, `96k 24` is for USB, but not for bluetooth.

I think, why outputting to the phone instead of a bluetooth headset is that there isn't any "bluetooth" audio module on your phone. Your phone probably uses only "a2dp hardware offload" driver for outputting to bluetooth headsets. In this case, audio data is usually resampled twice at a mixer of the "a2dp hardware offload" driver (default: 48kHz & 16 or 24bit) and at the bluetooth driver (LHDC: 96kHz & 24bit).
If so, you can change the sample rate of the former mixer by specifying "--offload 96k 24" option.

For USB, the special mediaplayer probably uses "AAUDIO" feature of recent Android OS's. You can raise the resampling quality of the OS mixer (AudioFlinger) to a very mastering quality (no resampling distortion in a real sense). See "extras/change-resampling-quality.sh".

Andrea-lyz, 2022-11-18:
> `bypass-offload-safer` changes the sample rate of a mixer for USB only and disables "a2dp hardware offload" for bluetooth headsets so that the sample rate of a mixer for bluetooth can be changed [...]

Thank you for your reply.
I tried `--offload 96k 24` on the bluetooth headset as you did, but still had the problem that the headset no longer made sound, but sounded from the phone. Then I tried `---safe/--safest 96k 24`, and the typical example of automatic setting `96k 24`, again with various problems, such as no sound at all or sound from the phone. This means that the audio is only decoded by the LHDC driver? That is, there is no quality loss? Or is there something else I can try?

yzyhk904, 2022-11-19:
Did you turn off bluetooth before executing my script? Unless turning off bluetooth, the audioserver may not function correctly.

If you turned off bluetooth, I think your phone needs some more special configurations. I made template configurations in my script so that they would match devices as much as possible by enabling only minimum common functions. Try to make a new template from the stock "audio_policy_configuration.xml" perhaps in "/vendor/etc" folder to replace with "offload_template.xml" in "templates" folder.

Andrea-lyz, 2022-11-19:
> Did you turn off bluetooth before executing my script? Unless turning off bluetooth, the audioserver may not function correctly.
> 
> If you turned off bluetooth, I think your phone needs some more special configurations. I made template configurations in my script so that they would match devices as much as possible by enabling only minimum common functions. Try to make a new template from the stock "audio_policy_configuration.xml" perhaps in "/vendor/etc" folder to replace with "offload_template.xml" in "templates" folder.
> 
> Edit: By some chance, this is because of special selinux enforcing on miui. If so, try `setenforce 0` before using my script. You can revert `setenforce 1` after the script execution.

Unfortunately, whether Bluetooth is off or on, or I replace` offload_template.xml` in the script with the native `audio_policy_configuration.xml`, it still doesn't work. I also tried turning SELinux off completely aka `setenforce 1/0` (I understand that this will allow some modules to work properly, like ViPER4Android FX), but it still has the problem of only sound coming from the phone.
Maybe I can hand you something for further analysis of the problem?
Thanks for your work!

yzyhk904, 2022-11-20:
If replacing the template with the stock one unmodified, the audioserver must work the same as before. Because the audioserver loads the same configuration xml.

By the way, your configuration file is common for a2dp hardware offloading and doesn't have capability for LHDC (only for LDAC, AAC, SBC, aptX and aptX HD). Can you use LHDC on the native stock ROM?

I think generated configuration files are corrupted with some reason. My scripts are written for "ash" but not "bash". Don't you use  "bash" instead of "ash" (standard shell on Androids; in "toybox" or "busybox")?
I've heard "termux" uses "bash" and some magisk module replaces the "ash" with "bash".

Andrea-lyz, 2022-11-20:
> If replacing the template with the stock one unmodified, the audioserver must work the same as before. Because the audioserver loads the same configuration xml.
>
> By the way, your configuration file [...]

Are you saying that I need to additionally modify `audio_policy_configuration.xml`?
I don't quite understand how to modify it.
For LHDC, I'm sure I'm using it in the native stock ROM (screenshot), my headset only supports `LHDC_V4` and is not forward compatible（LHDC_V1/2/3）, maybe it has extra stock?
As for bash and ash, I'm not sure (maybe bash?), I used termux and Tasker to run your script.And busybox installed in magisk.
[image]
[image]

yzyhk904, 2022-11-20:
I think my script cannot run correctly because of termax interference. Try uninstalling termax and tasker.
I recommend using sManager (ad version).

Andrea-lyz, 2022-11-27:
> I think my script cannot run correctly because of termax interference. Try uninstalling termax and tasker. I recommend using sManager (ad version).

I don't think it's a problem with termax and tasker, it's in Script Manager-SManager (I think I downloaded it right?) The same problem as above, the sound always changes from the speaker on the phone side and no sound from the bluetooth headset.
[image]
This is the log displayed by dumpsys-filtered.sh when the Bluetooth headset is playing normally through mediaplayer
[image]

closed 2023-01-08 by yzyhk904