yzyhk904/USB_SampleRate_Changer #17: "audio_policy_configuration.xml doesn't exist!"

Oxi-nya, 2023-01-20:
Device: OnePlus 9 Pro with crDroid 13 (LineageOS based). I'm also using 2 modules: "DRC remover" and "USB Samplerate unlocker"

For some reason executing the script with USB DAC connected only throws errors (With parameters it seems to also throw error)

`~ $ sh /sdcard/script/USB_SampleRate_Changer.sh
/sdcard/script/USB_SampleRate_Changer.sh: 7: /sdcard/script/functions3.shlib: Syntax error: "(" unexpected
~ $ su -c sh /sdcard/script/USB_SampleRate_Changer.sh
overlaying target ("/vendor/etc/audio/sku_lahaina_qssi/audio_policy_configuration.xml") doesn't exist!`

Error didn't occur on Nameless AOSP

yzyhk904, 2023-01-21:
First of all, `"(" unexpected` means you used "bash " instead of standard "ash". If you installed some "binutils" module and/or "termux", please uninstall them. They aren't Android standard. This tool is developed for Android standard environments, i.e., toybox and busybox commands (not compatible to binutils and termux ones).

If those XML configuration files really exist, I think Magisk mount name space hasn't been changed to "global" (default is "inherited").

Oxi-nya, 2023-01-21:
As for XML configs, there's only this. In /etc/ there's no usb_audio_policy_configuration.xml
[image]

Command also cannot execute from Script Manager. In Nameless AOSP I could execute the command with any parameters from termux or from Script Manager

yzyhk904, 2023-01-22:
There isn't "usb_audio_policy_configuration.xml" on this device, so this command cannot overlay any modified usb audio policy on it and fails. This is normal behavior. Other many devices have "usb_audio_policy_configuration.xml" and this command would succeed on them. However, audioserver on many devices cannot load this configuration because their primary configuration doesn't include the configuration file but has its contents in itself.

I think "termux" interferes with my command. Please uninstall "termux" completely. At least, "bash" cannot run my command and some command supplied from "termux" aren't compatible to toybox and busybox commands. Finally, did you change magisk "mount name space" to "global" in magisk settings?

Oxi-nya, 2023-01-23:
I've verified the script outputs the same error, when "Mount space name" is set to "Global", both for termux and Script Manager. Without termux, Script Manager doesn't properly execute the script, with the same error

yzyhk904, 2023-01-24:
I think there could be invisible "/vendor/etc/audio/sku_lahaina_qssi/audio_policy_configuration.xml" by any root app, but Audioserver can read it on this crDroid.

If there wasn't "/vendor/etc/audio/sku_lahaina_qssi/audio_policy_configuration.xml", you couldn't hear any sound through USB DAC's after clean installation of the crDroid.

I made a modified USB_SampleRate_Changer to skip the target XML file checking, try
[USB_SampleRate_Changer.zip](https://github.com/yzyhk904/USB_SampleRate_Changer/files/10487438/USB_SampleRate_Changer.zip)

Oxi-nya, 2023-01-24:
Modified script executed properly, thank you very much!

yzyhk904, 2023-01-25:
I think the invisible file is caused by a bug of the crDroid. Probably the mount point should be "/vendor/etc/audio/sku_lahaina/audio_policy_configuration.xml" like stock ROM's.

Anyway, this fix will be included in the next release.