yzyhk904/USB_SampleRate_Changer #19: Sintax error: unexpected 'fi'

hibikidesuuu, 2023-02-14:
I ran the script on smanager but what i get is this. Please help.
Magisk version: `25.2`
OS: `crDroid a11`

`# /system/bin/sh /storage/emulated/0/USB_SampleRate_Changer/USB_SampleRate_Changer.sh --auto
/storage/emulated/0/USB_SampleRate_Changer/USB_SampleRate_Changer.sh[324]: exite6is_: inaccessible or not found
/storage/emulated/0/USB_SampleRate_Changer/USB_SampleRate_Changer.sh[325]: syntax error: unexpected 'fi'`

yzyhk904, 2023-02-14:
Although I've checked my script with "sh -n" and "sh -x" on my environments, there is no syntax error, fail and anything strange. I think you used probably another sh like "bash". My script is compatible with ash in both "toybox" and "busybox", but not other sh's. Or you didn't use the global mount name space that can be set in the Magisk manager settings.

hibikidesuuu, 2023-02-14:
I already have it set to global mount. I still get the same error. All i have to do is run the script right or am i doing something wrong?

[image]
[image]
[image]

yzyhk904, 2023-02-15:
Do you use a OnePlus device?  I've found and fixed a bug for OnePlus devices only.
Try a following candidate to release after deleting the previous one completely.

[USB_SampleRate_Changer-V2-6-3.zip](https://github.com/yzyhk904/USB_SampleRate_Changer/files/10744792/USB_SampleRate_Changer-V2-6-3.zip)

By the way, sManager cannot change a shell command specified at Shebang. If you changed "/system/bin/sh" to another, please revert it to its original.

closed 2023-03-08 by yzyhk904