yzyhk904/USB_SampleRate_Changer #21: It will cause the hands-free call to be disabled, is there a solution?

yejinson, 2023-04-12:
I just need to change the sample rate of Bluetooth.

yzyhk904, 2023-04-13:
Tell me your situation more and precisely. Is there any relation between your comment and the title?

I found a possible bug for the configuration of the primary input mixer. Do you use "--bypass-offload-safer" option or default (no option)? If so, try "--bypass-offload 48k 16".

This script cannot change the sample rate of Bluetooth which is independent of the OS mixer. You may change it via "Developer Options" when your Bluetooth device is connected.

yejinson, 2023-04-14:
> Tell me your situation more and precisely. Is there any relation between your comment and the title?
>
> I found a possible bug for the configuration of the primary input mixer. Do you use [...]

I tried all the options, sorry I was wrong, I mean change the system output sample rate, when I use the --bypass-offload option, the player cannot use Bluetooth, it will make sound from the built-in speaker, use --offload, the playback is normal, but the hands-free will be disabled.

yejinson, 2023-04-15:
> What device and OS are you using? It seems that you are using a recent Xperia or OnePlus phone and its stock OS. And what app are you using for telephone communication (VoLTE? VoWiFi? VOIP?)

Realme10pro,realmeui4.0(android13) ,VoLTE.

yzyhk904, 2023-04-16:
Stock OS's aren't supported because I'm using AOSP and LineageOS based custom ROM's only on limited devices. Although I looked at the contents of your firmware, I think "--offload 48k 16" must work.

I updated the latest released zip that fixed few bugs related hands-free, try again if you like.
Perhaps "--bypass-offload-safer" mode may work if you edit "templates/bypass_offload_safer_template.xml" so that `<module name="bluetooth" halVersion="2.0">` is replaced with `<module name="bluetooth_qti" halVersion="2.0">`.  

The "bluetooth" HAL driver won't work, but the (a2dp hardware offload) primary HAL driver can handle bluetooth audio. The "bluetooth_qti" HAL driver (another bluetooth driver of Qcomm) may work, but I'm not sure.

yejinson, 2023-04-16:
> Stock OS's aren't supported because I'm using AOSP and LineageOS based custom ROM's only on limited devices. Although I looked at the contents of your firmware, I think "--offload 48k 16" must work.
> 
> I updated the latest released zip that fixed few bugs related hands-free, try again if you like. Perhaps "--bypass-offload-safer" mode may work if you edit "templates/bypass_offload_safer_template.xml" so that `<module name="bluetooth" halVersion="2.0">` is replaced with `<module name="bluetooth_qti" halVersion="2.0">`.
> 
> The "bluetooth" HAL driver won't work, but the (a2dp hardware offload) primary HAL driver can handle bluetooth audio. The "bluetooth_qti" HAL driver (another bluetooth driver of Qcomm) may work, but I'm not sure.
> 
> Edit: this script works file on custom ROM's for bluetooth SCO (Hands-free) even if specifying various sampling rates, e.g. 44.1kHz, 48kHz, 96kHz.

I tried the new version, neither "--offload 48k 16"  nor "--bypass-offload-safer"  with replace template "bluetooth_qti" will work（Hands-free).

yzyhk904, 2023-04-16:
I have no idea now. Probably Realme UI specific something prevents bluetooth SCO.

I thought before "--offload 48k 16" would work at least if you executed the script after closing bluetooth connection. But it failed.

yejinson, 2023-04-16:
> I have no idea now. Probably Realme UI specific something prevents bluetooth SCO.
> 
> I thought before "--offload 48k 16" would work at least if you executed the script after closing bluetooth connection. But it failed.

I'm not sure if you understand what I'm saying, the hands-free failure refers to the inability to use the internal speaker for calls, not Bluetooth. There is a problem with the internal speaker, it cannot be used for calls, but it can play music.

yzyhk904, 2023-04-17:
Do you mean you cannot use any telephone app with correct volume control after executing this script?
If so,  because your phone doesn't load any correct volume table.

This script try to load the volume table from Android standard places, i.e., "/vendor/etc/audio_policy_volumes.xml" and "/vendor/etc/default_volume_tables.xml". But Oppo brand and Xperia devices use other device specific places, e.g. your phone uses "/vendor/etc/audio/sku_holi/audio_policy_volumes.xml".

Try to replace "/vendor/etc/audio_policy_volumes.xml" and /vendor/etc/default_volume_tables.xml" with "/vendor/etc/audio/sku_holi/audio_policy_volumes.xml" and /vendor/etc/audio/sku_holi/default_volume_tables.xml" in templates XML files.

I attach an "offload_direct_template.xml" for your phone specific use for replacing. Replace and use with "--offload-direct 48k 16" option and options with another sample rate and bit depth.
[Realme-specific-offload_direct_template.zip](https://github.com/yzyhk904/USB_SampleRate_Changer/files/11248537/Realme-specific-offload_direct_template.zip)

yzyhk904, 2023-04-17:
What do you mean "handset"? Bluetooth earphones and the like? An internal speaker at a low volume? A USB headset? And what's the "hand-free" button? "speaker" (an internal speaker at a high volume) button in an app?

Is there "/vendor/etc/audio/sku_holi/audio_policy_volumes.xml" on your phone?

yejinson, 2023-04-17:
> What do you mean "handset"? Bluetooth earphones and the like? An internal speaker at a low volume? A USB headset? And what's the "hand-free" button? "speaker" (an internal speaker at a high volume) button in an app?
> 
> Is there "/vendor/etc/audio/sku_holi/audio_policy_volumes.xml" on your phone?

Handset means earpiece, "hand-free" means an internal speaker at a high volume, yes, there is "/vendor/etc/audio/sku_holi/audio_policy_volumes.xml" on my phone.

yzyhk904, 2023-04-18:
I checked various audio configurations on several phones here, but all work well.
I think your OS prevents to change the audio configuration completely.
The template file I attached is the same as the audio policy configuration file on your phone (except parameterization of sample rate and bit depth).

If you execute the script with "-r" option after executing it with other options, you can revert back to the original audio state as your OS doesn't prevent it. Can you try this?

yejinson, 2023-04-18:
> I checked various audio configurations on several phones here, but all work well. I think your OS prevents to change the audio configuration completely. The template file I attached is the same as [...]

Hands-free returns to normal after execute the script with "-r" option.

yzyhk904, 2023-04-19:
Um,  I cannot understand what happened.

I attach "offload_direct_template.xml" for your phone literally copied from the audio_policy_configuration.xml file in a Realme 10 Pro 5G firmware. After replacing the template file in "templates" folder with this, my script with "--offload-direct" option will change the audio configuration to the same as the original state.

If this succeeds and you modify the replaced "offload_direct_template.xml" to be parameterized the same as other template files with %-embraced parameters, you can change the sample rate and the bit depth of bluetooth earphones, USB DACs, and internal speakers.
[Realme-offload_direct_template.zip](https://github.com/yzyhk904/USB_SampleRate_Changer/files/11273304/Realme-offload_direct_template.zip)

yejinson, 2023-04-20:
> Um, I cannot understand what happened.
>
> I attach "offload_direct_template.xml" for your phone literally copied from the audio_policy_configuration.xml file in a Realme 10 Pro 5G firmware. After [...]

I replaced the "offload_direct_template.xml" file and used "--offload-direct", still failed. I checked the file, which is consistent with the file in the /vendor/etc/audio/ path on my mobile phone. In addition, there is another file with the same file name but different content in the /vendor/etc/audio/sku_holi/ path, as invalid, attached files.
[audio_policy_configuration.zip](https://github.com/yzyhk904/USB_SampleRate_Changer/files/11286685/audio_policy_configuration.zip)

yzyhk904, 2023-04-20:
`/vendor/etc/audio/sku_holi/audio_policy_configuration.xml` is a dummy (a little bit general one for the holi board family). `/vendor/etc/audio/audio_policy_configuration.xml` is the true configuration file. Perhaps it is overlaied on `/vendor/etc/audio_policy_configuration.xml`

Do you have `.config` file in the USB_SampleRate_Changer folder?  "PolicyFile" is the filename which will be loaded by the audioserver. Isn't this `/vendor/etc/audio_policy_configuration.xml`?

yejinson, 2023-04-21:
> `/vendor/etc/audio/sku_holi/audio_policy_configuration.xml` is a dummy (a little bit general one for the holi board family). `/vendor/etc/audio/audio_policy_configuration.xml` is the true [...]

No, Policy File is "/my_product/etc/audio_policy_configuration.xml", I change it to "/vendor/etc/audio_policy_configuration.xml", and the hands-free can work now, but neither "--bypass-offload 44.1k 24" nor "--offload 44.1k 24" can change the system sampling rate. By the way, I don't understand the difference between these options, which option has the best sound quality for Bluetooth?

yzyhk904, 2023-04-21:
I found new audio configuration file in `/my_product/etc`. ColorOS loads this file, but not `/vendor/etc/audio/audio_policy_configuration.xml` (probably for OxygenOS; not used).

Remove the `.config` file and replace `offload_direct_template.xml` with attached new one, then execute this script with `--offload-direct 48k 24` option. I think this could succeed. 

I think "hands-free" can succeed only with 24bit mode.
[Realme-my_product-offload_direct_template.zip](https://github.com/yzyhk904/USB_SampleRate_Changer/files/11297456/Realme-my_product-offload_direct_template.zip)

By the way, `bypass-offload` uses AOSP's "bluetooth" driver and `offload` uses Qcomm's a2dp harwdware offload driver. The former is better in audio quality (needing less resampling). "bluetooth_qti" is Qcomm's old driver.

yejinson, 2023-04-22:
> I found new audio configuration file in `/my_product/etc`. ColorOS loads this file, but not `/vendor/etc/audio/audio_policy_configuration.xml` (probably for OxygenOS; not used).
>
> Remove the [...]

Problem solved, I really appreciate your help.

closed 2023-04-22 by yejinson