yzyhk904/USB_SampleRate_Changer #22: LDHC

Mario6425, 2023-08-28:
LDHC
Is it possible to use the script with a LDHC codec? When I use my oneplus buds 2 pro, I can only play sound under the AAC or SBC codec. Is there a way to do this?

yzyhk904, 2023-08-29:
How do you use my script?  "--offload" mode or automatic selection? Do your phone support LHDC codec?

I don't have any LHDC capable earphones. However you may use LHDC codec by some chance if the codec is supported on your phone and "--offload 48k 24" or the like option is used.

Mario6425, 2023-08-29:
Hey, thanks for the quick feedback.
What I wanted to say, that your script is really great. Even EQ like maxmpz his poweramp API become superfluous.

To your question: I use your script over various modes from automatic to all other options, depending on what just fits the device. Depends if I use my V8 module with ciem, Fiio BTR5 via Usb Dac in the car (ultra good) or Qudelix 5k. 
Everything works fine.

But with the oneplus Buds pro 2 all scripts from the extra templates work with LHDC. But if you use the Samplerate-Changer, can only listen to music via AAC or SBC. I can play strangely even with AAC 192k 24 bit, of course comes via Bluetooth reduced.
I have really tried all the options 
With the BTR5 over or other LDAC devices everything works: 
Is LDHC perhaps encoded differently

yzyhk904, 2023-08-30:
There are three bluetooth audio drivers, i.e., "bluetooth" (AOSP's driver), "offload" (Qcomm's bluetooth offload driver) and "bluetooth_qti" (Qcomm's legacy bluetooth driver). "offload" and "bluetooth_qti" can/may support LHDC codec.

Did you mean you used "--offload" option with USB_SampleRate_Changer? Then, if you use ColorOS which handles LDHC by "bluetooth_qti" audio driver, there is one more chance for LHDC.

Try "--force-bluetooth-qti" option with USB_SampleRate_Changer. This option forces my script to use "bluetooth_qti" driver if possible on your phone. Check it by "getConfig.sh" command.

Mario6425, 2023-09-01:
These are the commands I use plus when I switch to permissive in selinux to apply the jitter reducer.

sh /sdcard/USB_SampleRate_Changer/USB_SampleRate_Changer.sh --offload 44k 16

sh /sdcard/USB_SampleRate_Changer/extras/change-resampling-quality.sh 194 520 100

 --force-bluetooth-qti does not work there is no file for it in the script. Or I set the command wrong.

sh /sdcard/USB_SampleRate_Changer/USB_SampleRate_Changer.sh --force-bluetooth-qti 44k 16

/sdcard/USB_SampleRate_Changer/USB_SampleRate_Changer.sh[227]: [: missing

Either about the drivers yet, or I make a screenshot of the getConfig.sh

yzyhk904, 2023-09-01:
I fixed this bluetooth-qti bug. And "--offload" is wrong.

"--legacy" for Xiaomi devices and "--force-bluetooth-qti" for ColorOS. LHDC may be supported by these modes.

Try this fixed zip.
[USB_SampleRate_Changer-v2.8.1-testing.zip](https://github.com/yzyhk904/USB_SampleRate_Changer/files/12497699/USB_SampleRate_Changer-v2.8.1-testing.zip)

Mario6425, 2023-09-02:
Thanks for your help and the script bro. 

I have tested it completely. 
--force-bluetooth-qti is selectable and the LDHC codec is running, but there is no change in the sample rate(even with --legacy). 

All other menu items in the script work, but only with AAC. By the way, I use a Oneplus 11 (cph2449) with oxygen os 13.1.

yzyhk904, 2023-09-03:
You may change the sample rate in the developer settings (in the bluetooth section). There are two re-samplers in an audio path. One in the OS mixer is automatically adapted to bluetooth transmitter settings in this situation. The other is in a bluetooth transmitter. You can change this via the developer settings.

closed 2023-09-29 by yzyhk904