yzyhk904/USB_SampleRate_Changer #28: Audio speed and pitch halved with 192k 32, no audio with 384k 32

Epikest, 2024-02-11:
i tried the script on my Google Pixel 6 and when i change the sampling to 192k 32 any audio played is halved in playback rate and pitch. using 384k 32 there is no audio at all.

yzyhk904, 2024-02-12:
Pixel6's cannot replay tracks over 96kHz without usb_samplerate_unlocker (up to 192kHz for Pixel's) because of their platform configuration.

This is normal behavior.

Epikest, 2024-02-13:
> Pixel6's cannot replay tracks over 96kHz without usb_samplerate_unlocker (up to 192kHz for Pixel's) because of their platform configuration.
> 
> This is normal behavior.

what is the benefit that this script brings to usb-samplerate-unlocker? changing via command line?

yzyhk904, 2024-02-14:
usb_samplerate_unlocker extends only the capability of the Pixel6's original sample rate range [8 ~ 96] kHz to [8 ~ 192] kHz. The sample rate of a USB DAC isn't changed at this point, usually 48kHz for 44.1kHz tracks and 96kHz for hires. ones.

This script can change the rate in [44.1 ~ 192] kHz range on the fly by executing it with options.

By the way, Tensor devices are poor in this point, but others are can be extended the rage to [8 ~ 768] kHz.

closed 2024-03-09 by yzyhk904