yzyhk904/USB_SampleRate_Changer #29: Requesting a acript to make 100 volume steps on the fly

Cico90, 2024-03-10:
Hello,

I was using your hifi-maximizer magisk module. I liked the performance boost it got me ans the 100 volume steps. Now, I am just using this script for the performance boost but I need a way to change the volume steps to a 100. Is that possible with this scripts or do you have one just for that!?

You are very much appreciated!

yzyhk904, 2024-03-10:
Don't you like [Audio Misc. Settings](https://github.com/Magisk-Modules-Alt-Repo/audio-misc-settings) (that changes the number of volume steps to 100 and other things)?

Cico90, 2024-03-10:
But Audio-Misc-Settings disables the effects framework and I don't want that ( vlc dosn't have audio with effects turned off ).

Cico90, 2024-03-10:
I tried re-enabling the effects framework using this script : jitter-reducer_++effect.sh
But there is still no sound. Do I have to restart the effects framework service or the audio service or something!?
And how to do it if so??

yzyhk904, 2024-03-11:
The script restarts the audioserver and audio HAL's automatically after restarting the effects framework. Did you enable the framework just after rebooting (before VLC starting)? 

If so, I guess the framework is nothing to do with this issue. Try "change-resampling-quality_Reset.sh" and "change-usb-period_Reset.sh" that revert the resampling and the USB period of Android OS to its default settings.

Cico90, 2024-03-11:
It took a little bit of time to work so it didnt work right away after I re-enabled the framework. I didn't notice it till a long time after.
But it is working.
👍👍👍