yzyhk904/USB_SampleRate_Changer #33: Bit depth and sample rate won't change

xiiyuki, 2024-08-17:
Hi, I will try to explain it briefly

For some reason, while using Moondrop Dawn Pro on OnePlus 9 Pro (Nameless AOSP, android 14, KernelSU with Global inheritance for script manager and termux), in apps like Spotify, MPV, Android chooses some format that in combination with it's resampling, causes very noticeable sound cut-off when it's near silent. Poweramp has Hi-Res Output, that is able to change sampling rate and bit depth of the DAC (which it indicates with RGB LED), which mitigates the issue instantly, by setting proper bit depth and sample rate manually.

I tried to resolve it:
1. By installing Audio misc. settings

xiiyuki, 2024-08-17:
I've got no idea why it published mid-writing, I'll edit original message in a few minutes

yzyhk904, 2024-08-17:
First of all, I think KernelSU dosn't have the global mount mode, but does have "su --mount-master" option. Please use scripts under "Usage_samples" with "mixplorer" or something.

Poweramp uses only "direct_pcm" modes of the default audio policy on your phone, but spotify app does a usual "deep_buffer" mixer whose sample rate is fixed at 48 kHz.

I guess you could change the sample rate and bit depth of the default mixer by scripts under "Usage_Samples/USB".

closed 2024-08-28 by yzyhk904