yzyhk904/USB_SampleRate_Changer #34: Pixel 9 pro xl ("/vendor/etc/audio_policy_configuration.xml") doesn't exist!

g1nt0, 2024-10-31:
SampleRate_Changer_44_32.sh; exit &>/dev/null             <
overlaying target ("/vendor/etc/audio_policy_configuration.xml") doesn't exist!

[Process completed (error 1) - press Enter]

yzyhk904, 2024-10-31:
What's the SampleRate_Changer_44_32.sh? Try scripts under "Usage_Samples".

g1nt0, 2024-10-31:
same(

USB/USB_48k_24.sh; exit &>/dev/null            <
overlaying target ("/vendor/etc/audio_policy_configuration.xml") doesn't exist!

[Process completed (error 1) - press Enter]

g1nt0, 2024-10-31:
:/data/data/com.termux/files/home # setenforce 0
:/data/data/com.termux/files/home # getenforce
Permissive
same result(

~ $ su
:/data/data/com.termux/files/home # ls -lZ /vendor/et>
ls: /vendor/etc/audio_policy_configuration.xml: No such file or directory
1|:/data/data/com.termux/files/home #

yzyhk904, 2024-10-31:
The file must be there, and I think it may not happen.

I guess "Termax" might do something wrong. Please use Mixplorer (free version) to execute "Usage_Samples". See README.

g1nt0, 2024-10-31:
[image]
this file is not here

g1nt0, 2024-10-31:
I can't find this file anywhere at all(

g1nt0, 2024-10-31:
[image]

g1nt0, 2024-10-31:
@yzyhk904 
`adb shell
su
find / -print > `/sdcard/full_catalog.txt`
[image]

yzyhk904, 2024-11-01:
I researched the audio system of Pixel9, and found it had only AIDL, but not HIDL (audio policy configuration by XML). This script cannot run on AIDL only devices like Pixel 9 series (built for their AIDL only configurations).

I will add codes for excepting Pixel 9 series and other AIDL only devices, and add info in README.

g1nt0, 2024-11-01:
could this change with software updates or if the device was released like this right away, will it be the same in a row?

yzyhk904, 2024-11-01:
Same. This script can support only HIDL based devices. If there is a compatibility command converting HIDL to AIDL, I may support AIDL devices future.

By the way, there is a build option in Pixel 9 series firmware sources for HIDL instead of AIDL. If you build the firmware for HIDL, you could use this script.

closed 2024-11-09 by yzyhk904