yzyhk904/USB_SampleRate_Changer #36: audio_policy_configuration.xml not found in /vendor/etc

buratei, 2024-12-12:
Hi, I am getting the error message in the title when applying the scrip with default parameters (simply sh /sdcard/USB_SampleRate_Changer/USB_SampleRate_Changer.sh). The device is Nexus 7 2013 LTE, and the OS it has is LineageOS 14.1. Maybe it is located in another directory? Would it work if I locate the file, temporarily put it in /vendor/etc/ , run the script and then relocate it to wherever I found it in the first place? Than you.

yzyhk904, 2024-12-13:
Did you use the latest script? It doesn't produce such error messages (but something similar). And LineageOS 14.1 is Android 7.1 that this script cannot support. Your device doesn't have audio_policy_configuration.xml, but audio_policy.conf instead.

If you upgrade its ROM over or equal to LineageOS 15.1, you could use this script.

buratei, 2024-12-13:
I see, thank you for the reply.
If I want to stay on LoS 14.1, would it be possible to achieve similar effect by manually editing audio_policy.conf?

yzyhk904, 2024-12-13:
I don't know things before Android 8.1, but old devices seems to be changed very little by a configuration.

closed 2024-12-13 by yzyhk904