yzyhk904/USB_SampleRate_Changer #39: Best setting for Spotify

StephySARS, 2025-04-19:
My phone is OnePlus 7 pro with Crdroid 11 (android 15).
Mainly use Spotify streaming to BT headphones with LDAC 44.1k/16bit setting. 
Spotify is streaming ogg in 44.1k sample rate according to the official website. 
But I found that in other issue mentioned "spotify app does a usual "deep_buffer" mixer whose sample rate is fixed at 48 kHz."
So what's the best setting to avoid resampling in my case? Currently I am using "change-resampling-quality.sh --bypass". Thank you very much for making the scripts.

yzyhk904, 2025-04-20:
I think your phone uses a hardware offload BT driver which is considerably worse in audio quality than an AOSP BT driver. 

First of all, execute "Usage_Samples/USB/USB_auto_max.sh" to switch to the AOSP BT driver and to avoid the deep_buffer mixer. Then, you can use an automatic mixer to match to your LDAC settings dynamically.

I think your best LDAC settings are 96kHz & 32bits & best quality (990kbps) and the best resampling is "Usage_Samples/Resampling/change-resampling-quality_cheapies_44kHz.sh" whose audio quality is far better than that of the over-sampling filter in any audio device as far as I know. Avoiding resampling is no meaning if you can use a far better resampler than the over-sampling filter in your device.

If the AOSP BT driver wouldn't work on your phone anyway, use 48kHz & 24bits & best quality and the cheapies-44kHz.sh.

StephySARS, 2025-04-20:
Thank you for the reply.
I ran the dumpsys.sh and got the following info:

Output thread 0xb4000079a33ba880, name AudioOut_23D, tid 17817, type 0 (MIXER):
  Sample rate: 44100 Hz
  HAL format: 0x1 (AUDIO_FORMAT_PCM_16_BIT)
  Output devices: 0x80 (AUDIO_DEVICE_OUT_BLUETOOTH_A2DP)
  Timestamp stats: n=3615 disc=2 cold=2 nRdy=10 err=213 rate=0.999984 jitterMs(ave=0.000561254 std=0.95171 min=-3.49001 max=5.74951) localSR(44100.2, 1.22805e-07) correctedJitterMs(ave=-7.62709e-05 std=0.037686 min=-0.606398 max=1.17331)
  AudioStreamOut: 0xb40000782641e520 flags 0 (AUDIO_OUTPUT_FLAG_NONE)

Ran the bluetooth-hal_Status.sh got:

aosp ("bluetooth" HAL) : enabled

StephySARS, 2025-04-20:
Thank you for the reply.
I ran the dumpsys.sh and got the following info:

Output thread 0xb4000079a33ba880, name AudioOut_23D, tid 17817, type 0 (MIXER):
Sample rate: 44100 Hz
HAL format: 0x1 (AUDIO_FORMAT_PCM_16_BIT)
Output devices: 0x80 (AUDIO_DEVICE_OUT_BLUETOOTH_A2DP)
Timestamp stats: n=3615 disc=2 cold=2 nRdy=10 err=213 rate=0.999984 jitterMs(ave=0.000561254 std=0.95171 min=-3.49001 max=5.74951) localSR(44100.2, 1.22805e-07) correctedJitterMs(ave=-7.62709e-05 std=0.037686 min=-0.606398 max=1.17331)
AudioStreamOut: 0xb40000782641e520 flags 0 (AUDIO_OUTPUT_FLAG_NONE)

Ran the bluetooth-hal_Status.sh got:

aosp ("bluetooth" HAL) : enabled

Info above means I successfully switch to the AOSP BT driver and to avoid the deep_buffer mixer, right?

So it's no point to set the LDAC to 44.1/16@909kbps at all. Always use the 96/32@990k even if the sample rate of most of the audio files I have are 44k/48k?

yzyhk904, 2025-04-20:
Yes, you've already switched to the AOSP BT driver successfully and avoided the deep_buffer mixer (48kHz fixed rate). The best LDAC settings are 96kHz & 32bits & 990kbps (or best-effort) now. If you set 44.1kHz (and 16 bits), audio quality is getting worse then by the over-sampling filter for 44kHz input in earphones.

However, the best resampling parameters (i.e., low-pass filters) for 44k and 48k are different. You can find  scripts for switching between them in "Usage_Samples/Resampling" (_cheapies_44kHz.sh and _cheapies_48kHz.sh).

closed 2025-04-20 by StephySARS